wuphilipp/gello_mechanical #4: About 3D Printed Parts

ruanjinchen, 2024-01-25:
Is it convenient for you to provide the model in SolidWorks part format, because we would like to modify the model you posted, thank you very much！🙏🙏

wuphilipp, 2024-01-27:
please drop me an email!

closed 2024-01-27 by wuphilipp

ruanjinchen, 2024-01-27:
ruanjinchen@foxmail.com
THANKS!!!!!🫡

ShangQingLiu, 2024-03-21:
Hi, Thank you for your brilliant work, I would also like to have the model in SolidWorks part format which for creating my own one for Diana-7 Arms.  Here is my mail: shang-ching.liu@uni-hamburg.de

weixiang-smart, 2024-06-27:
Thank you for your excellent work! I would like to design a controller for UR3 in SolidWorks. I would appreciate it if you could share the models. Thank you so much in advance. My email is liangweixiang89@gmail.com

Cheney-36, 2024-07-23:
Thank you for sharing the content. I would like to modify the UR5 model to fit my installation base. If you could share the model, I would greatly appreciate it. My email is cy1440224730@gmail.com

wuphilipp, 2024-07-24:
Hi, thanks for you interested in this project! 

In the future for others please email me directly (you can find it on my website https://wuphilipp.github.io/) since I don't check this thread (its already closed).

Have just sent out the files to those of you who asked here.

Have fun and looking forward to what you all build!

dohu337, 2025-12-04:
Thank you for your brilliant work, I would also like to have the model in SolidWorks part format which for creating my own one for Franka Arms. Here is my mail: ddh05@naver.com
And I think Franka L1 part is broken. Can you upload new revision?