wuphilipp/gello_mechanical #12: Replace M288 Servos with M077 Servos. Servos are only being used for their encoders, so does their torque even matter?

jmcoholich, 2024-07-08:
We are planning to build the Gello device for Franka, which requires 7x M288 servos. 

The DYNAMIXEL website states that there is a 3-4 lead time for these: https://www.robotis.us/dynamixel-xl330-m288-t/
[image]

Is it possible to replace the M288s with M077 servos, which currently have no such lead time? It looks like the servos are identical in every aspect except their stall torque/rpm ratings. I believe this shouldn't even matter, since the servos are only being used for their encoders.

hygradme, 2024-07-23:
You can use M077 instead of M288. Sometimes,  I also use M077 for the joints near end effector to reduce the friction caused by higher reduction ratio gears and realized smooth joint movements.
But higher friction sometimes help more stable manipulation especially for the joints near the base of the arm where the effect of the gravity of the servos and other parts are higher and you need to resist against it. The higher friction reduces the gravity effect.

Also, you can try XC330-M288-T or XC330-M181-T if you have enough budgets.

jmcoholich, 2024-07-23:
Makes sense, thanks!

closed 2024-07-23 by jmcoholich

mehranraisi, 2026-10-12:
Hello Jeremiah,

I also have the same problem, could you please tell me how did you solve it? Is it possible to replace M288 with M077 servomotor?

Bests,
Mehran