wuphilipp/gello_mechanical #15: Requesting access to gello_mechanical source files (Solidworks)

chris24sahadeo, 2024-12-04:
(Mirror of email sent to phil80301@berkeley.edu)

Hey Phillip :)

The GELLO work is super impressive & helpful to the community :D

I'm interested in making a UR10 variant and, to that end, I'd like to request access to the gello_mechanical source files (Solidworks) as per the [repo's README](https://github.com/wuphilipp/gello_mechanical/tree/main).

Thanks!

yaotot, 2024-12-09:
i'am also very interested in  make the model with different UR robot. Could I also get  access to the gello_mechanical source files(Solidworks),

Thanks so much!

Michal-Rzaca, 2025-03-19:
It will be very helpful, please add solidworks files :)

Akumar201, 2025-03-19:
@chris24sahadeo @yaotot @Michal-Rzaca , where you able to get the solid-works file if yes would it be possible to share.

XueqiLi-Ai, 2025-03-22:
I want access to gello_mechanical source files too.

omccaffert-chwy, 2025-07-17:
Just adding to this thread! Planning to make a UR16e variant!

mguleken, 2026-09-02:
++ for the ur16e variant!